mcdeoliveira/rcpy #1: error: ‘rc_imu_read_mutex’ undeclared

MarkAYoder, 2017-05-17:
Looks like you have a great package here for working with the BeagleBone Blue.

I'm working with a Blue running "BeagleBoard.org Debian Image 2017-04-23"
I'm having trouble getting rcpy installed.
I've installed Robotics_Cape_Installer, ctrl and rcpy.
In ctrl I ran **sudo python3 setup.py install** and it appears to have installed without error.

However when installing in rcpy in get an error
$ **cd rcpy**
$ **sudo python3 setup.py install**
```
/usr/lib/python3/dist-packages/setuptools/dist.py:333: UserWarning: Normalizing '0.1a' to '0.1a0'
  normalized_version,
running install
running bdist_egg
running egg_info
writing dependency_links to rcpy.egg-info/dependency_links.txt
writing rcpy.egg-info/PKG-INFO
writing top-level names to rcpy.egg-info/top_level.txt
reading manifest file 'rcpy.egg-info/SOURCES.txt'
writing manifest file 'rcpy.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
running build_ext
building 'rcpy.mpu9250' extension
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-haFsHd/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c src/_mpu9250.c -o build/temp.linux-armv7l-3.5/src/_mpu9250.o
src/_mpu9250.c: In function ‘mpu9250_read’:
src/_mpu9250.c:335:26: error: ‘rc_imu_read_mutex’ undeclared (first use in this function)
     pthread_mutex_lock( &rc_imu_read_mutex );
                          ^~~~~~~~~~~~~~~~~
src/_mpu9250.c:335:26: note: each undeclared identifier is reported only once for each function it appears in
src/_mpu9250.c:338:25: error: ‘rc_imu_read_condition’ undeclared (first use in this function)
     pthread_cond_wait( &rc_imu_read_condition, &rc_imu_read_mutex );
                         ^~~~~~~~~~~~~~~~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

I don't know where **rc_imu_read_mutex** and **rc_imu_read_condition** are defined.

Any suggestions?

--Mark

mcdeoliveira, 2017-05-17:
For now you will need to compile the robotics cape library using my fork 

https://github.com/mcdeoliveira/Robotics_Cape_Installer

since James have not incorporated the mutexes in his version yet.

closed 2017-05-17 by mcdeoliveira

MarkAYoder, 2017-05-17:
That did it.  Many thanks for the quick response.

--Mark

https://github.com/mcdeoliveira/Robotics_Cape_Installer