mcdeoliveira/rcpy #3: error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

patrickpoirier51, 2017-07-30:
Brand new Image with rc_version 0.3.4

sudo apt-get install python3 python3-pip
sudo apt-get install python3-dev

Installed 1 Gb swap according to this:	
https://sheldondwill.wordpress.com/2013/12/14/beaglebone-black-ubuntu-adding-a-swapfile/

Still get the error:

root@beaglebone:/home/debian# sudo pip3 install rcpy
Downloading/unpacking rcpy
  Downloading rcpy-0.3.5.tar.gz
  Running setup.py (path:/tmp/pip-build-9ntf9he6/rcpy/setup.py) egg_info for package rcpy
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)

Installing collected packages: rcpy
  Running setup.py install for rcpy
    building 'rcpy._rcpy' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c src/_rcpy.c -o build/temp.linux-armv7l-3.4/src/_rcpy.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.4/src/_rcpy.o -lroboticscape -lrt -o build/lib.linux-armv7l-3.4/rcpy/_rcpy.cpython-34m.so
    building 'rcpy._mpu9250' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c src/_mpu9250.c -o build/temp.linux-armv7l-3.4/src/_mpu9250.o
    src/_mpu9250.c: In function ‘mpu9250_read’:
    src/_mpu9250.c:335:26: error: ‘rc_imu_read_mutex’ undeclared (first use in this function)
         pthread_mutex_lock( &rc_imu_read_mutex );
                              ^
    src/_mpu9250.c:335:26: note: each undeclared identifier is reported only once for each function it appears in
    src/_mpu9250.c:338:25: error: ‘rc_imu_read_condition’ undeclared (first use in this function)
         pthread_cond_wait( &rc_imu_read_condition, &rc_imu_read_mutex );
                             ^
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    Complete output from command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-9ntf9he6/rcpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-7snt5rok-record/install-record.txt --single-version-externally-managed --compile:
    running install

running build

running build_py

creating build

creating build/lib.linux-armv7l-3.4

creating build/lib.linux-armv7l-3.4/rcpy

copying rcpy/button.py -> build/lib.linux-armv7l-3.4/rcpy

copying rcpy/led.py -> build/lib.linux-armv7l-3.4/rcpy

copying rcpy/encoder.py -> build/lib.linux-armv7l-3.4/rcpy

copying rcpy/mpu9250.py -> build/lib.linux-armv7l-3.4/rcpy

copying rcpy/gpio.py -> build/lib.linux-armv7l-3.4/rcpy

copying rcpy/__init__.py -> build/lib.linux-armv7l-3.4/rcpy

copying rcpy/motor.py -> build/lib.linux-armv7l-3.4/rcpy

creating build/lib.linux-armv7l-3.4/test

copying test/test_led.py -> build/lib.linux-armv7l-3.4/test

copying test/test_mpu9250.py -> build/lib.linux-armv7l-3.4/test

copying test/test_encoder.py -> build/lib.linux-armv7l-3.4/test

copying test/test_motor.py -> build/lib.linux-armv7l-3.4/test

copying test/__init__.py -> build/lib.linux-armv7l-3.4/test

copying test/test_rcpy.py -> build/lib.linux-armv7l-3.4/test

running build_ext

building 'rcpy._rcpy' extension

creating build/temp.linux-armv7l-3.4

creating build/temp.linux-armv7l-3.4/src

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c src/_rcpy.c -o build/temp.linux-armv7l-3.4/src/_rcpy.o

arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.4/src/_rcpy.o -lroboticscape -lrt -o build/lib.linux-armv7l-3.4/rcpy/_rcpy.cpython-34m.so

building 'rcpy._mpu9250' extension

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c src/_mpu9250.c -o build/temp.linux-armv7l-3.4/src/_mpu9250.o

src/_mpu9250.c: In function ‘mpu9250_read’:

src/_mpu9250.c:335:26: error: ‘rc_imu_read_mutex’ undeclared (first use in this function)

     pthread_mutex_lock( &rc_imu_read_mutex );

                          ^

src/_mpu9250.c:335:26: note: each undeclared identifier is reported only once for each function it appears in

src/_mpu9250.c:338:25: error: ‘rc_imu_read_condition’ undeclared (first use in this function)

     pthread_cond_wait( &rc_imu_read_condition, &rc_imu_read_mutex );

                         ^

/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'

  warnings.warn(msg)

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-9ntf9he6/rcpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-7snt5rok-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-9ntf9he6/rcpy
Storing debug log for failure in /root/.pip/pip.log

mcdeoliveira, 2017-07-31:
Update roboticscape:

    sudo apt update
    sudo apt upgrade roboticscape

Unfortunately the release number for the library does not necessarily reflect the required changes.

patrickpoirier51, 2017-07-31:
Ok thanks, 
upgraded and installed no problem, just a little warning:
/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution                               option: 'python_requires'
      warnings.warn(msg)

Last word, it would be more evident  if rc_version would print all the detailed informations, here is what it shows during upgrade:
Unpacking roboticscape (0.3.4-git20170602-0rcnee1~jessie+20170602) over (0.3.4-git20170307-0rcnee2~bpo80+20170307)

Thanks again for the fast support, on this issue.

mcdeoliveira, 2017-08-01:
Happy to hear that you were able to solve the problem.

The warning is because distutils is outdated in the beaglebone and does not support the 'python_requires' clause yet.

As for the rc_version, I agree with you but this is a feature of the roboticscape library and not of the rcpy library and is beyond my control.

closed 2017-08-01 by mcdeoliveira