mcdeoliveira/rcpy #5: Analog in?

nezra, 2018-03-17:
Specifically battery voltage and DC?

What is needed to add these

BrendanSimon, 2018-04-11:
Yes, I'm interested in ADC inputs too.  e.g. to read distance sensors (optical, ultrasonic, etc).

What's the best way to read ADC inputs to use in conjunction with `rcpy`.  Ideally `rcpy` would implement an ADC module, but since it's not present is there another solution in the interim?

nezra, 2018-04-11:
my current solution, while horrible, was to write a small c program without
using the "state" function that just ran once, called the variables and
returned the variables via print. From there the python program runs it,
takes the output and converts/splits it into variables, which I then use.

I'd be wary of checking the led states for percentages as the battery
monitor assumes the battery depletes at a linear fashion instead of how it
really depletes. 25% battery  and 0 battery are less than a minute apart.

BrendanSimon, 2018-04-12:
The C libraries seem to be very comprehensive, and looking at the Python wrapper code it seems that it shouldn't be too difficult to add for the ADC module (and other missing modules too).

So I've taken the plunge (started a quick hacking investigation) to see if I can create the Python bindings to the low-level C functions.

BrendanSimon, 2018-04-14:
I've made some progress and my limited tests seem to suggest that it might work :)

You can find my fork/branch here:

https://github.com/BrendanSimon/rcpy/tree/bjs/adc_bindings

I've submitted a pull request here:

https://github.com/mcdeoliveira/rcpy/pull/7

mcdeoliveira, 2018-04-14:
Thank you Brendan! I already merged with master.

closed 2018-04-14 by mcdeoliveira