mcdeoliveira/rcpy #9: rcpy fails to install

cthomp1018, 2018-05-02:
debian@beaglebone:/var/lib/cloud9$ cat /etc/dogtag
BeagleBoard.org Debian Image 2018-03-05

debian@beaglebone:/var/lib/cloud9$ rc_version
0.4.2

Running latest image and version of Robotics Cape. I've attached a log of the failure. Occurs after running: sudo pip3 install rcpy

[rcpy installation failure.log](https://github.com/mcdeoliveira/rcpy/files/1965783/rcpy.installation.failure.log)

dakuenjery, 2018-05-05:
I have this problem too. Debian image 2017-09-21, rc_version 0.4.2

mcdeoliveira, 2018-05-05:
I am working on this problem and should have a solution in the next couple of days. The latest version of the roboticscape library breaks many calls and will need to be reworked.

mcdeoliveira, 2018-05-18:
Version 0.5 will install now with the latest version of the roboticscape. As I understand, there is still some issue between the kernel and the roboticscape library that is making all functions that rely on the pru, e.g. encoders, to fail. These should be solved soon. It will require no changes on rcpy.

closed 2018-05-18 by mcdeoliveira

singh-atul, 2018-12-21:
I tried installing version 5.0  with sudo pip install rcpy==0.5.0 , still i get the same error