mcdeoliveira/rcpy #12: "linux/gpio.h header not found" while installing libgpiod

gustavojcorrea, 2018-08-26:
I am currently using Ubuntu 16.04.5 LTS on Beaglebone Blue after following the [John Hopkins Edumip/ROS tutorial](https://dscl.lcsr.jhu.edu/home/courses/edumip_ros/). I continue to get stuck while trying to install libgpiod. Here is my process:

1. Burning image on SD card using etcher
2. Running `sudo apt-get update` and `sudo apt-get upgrade`
3. Following rcpy tutorial up until 
`./autogen.sh --enable-tools=yes --enable-bindings-python --prefix=/usr/local`
4. While running this command I get the following error
`checking sys/poll.h usability... yes
checking sys/poll.h presence... yes
checking for sys/poll.h... yes
checking linux/gpio.h usability... no
checking linux/gpio.h presence... no
checking for linux/gpio.h... no
configure: error: linux/gpio.h header not found (needed to build the library)`

I want to note that I have been successful in installing rcpy on Debian 9.3 however I ran problems when trying to install ROS. I tried the above process for Ubuntu. 

I would appreciate any help and advise on switching back to a specific Debian version!

mcdeoliveira, 2018-12-22:
This seems to be an issue with the ubuntu distribution not supporting libgpio yet. I am not sure what is the current status of support for ROS on debian.

closed 2018-12-22 by mcdeoliveira

dujunqiu, 2024-10-11:
please use  "find /opt/gcc-linaro-4.9.4-2017.01-x86_64_arm-linux-gnueabi/  -name  gpio.h", check your cross compile exist  gpio.h