mcdeoliveira/rcpy #14: Segmentation fault when making rcpy.motor calls

michaelmegliola, 2018-10-18:
I'm working on a BeagleBone Blue running the latest image (2018-10-7) suggested in the Strawson online manual, flashed to onboard storage.

I've started getting frequent crashes working in python, controlling dc motors.  A typical crash output is included below.

I can reproduce the issue reliably with steps like:
- Run code via Cloud9 to move my little robot around
- Click Stop in the IDE
- Run the code again through the IDE
- Segmentation fault

Most of the time, it crashes with a segmentation fault in Python.  Sometimes it will exhibit other erratic behavior, like move one motor but not the other, or run at 100% duty cycle when 20% was specified.  Sometimes, it runs fine.  It looks like something isn't getting properly cleaned up.

I was also able to duplicate the issue as follows, outside of the cloud9 IDE:
- Boot the Beagle and connect via ssh
- run 'rc_test_motors -s .2'    (which works fine) and cancel
- run 'rcpy_test_motors -s .2' (which does nothing) and cancel
- run 'rc_test_motors -s 2' again, resulting in a segmentation fault

Console output: 

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.114586] Internal error: : 1028 [#1] PREEMPT SMP ARM

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.233418] Process rc_test_motors (pid: 1884, stack limit = 0xd2026218)

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.240148] Stack: (0xd2027e18 to 0xd2028000)

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.244526] 7e00:                                                       00000001 00000320

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.252745] 7e20: 00001f40 d2027e7c db422900 c081df48 db422920 00001f40 00000000 dc6ce910

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.260962] 7e40: d2027e74 d2027e50 c081b72c c081df54 c1504dc8 00000000 dc60c994 db422900

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.269180] 7e60: 00000004 dc568b80 d2027eb4 d2027e78 c081be7c c081b618 00001f40 00009c40

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.277396] 7e80: 00001f40 00000000 00000001 77533b88 00000004 c081be00 dc6ce900 00000003

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.285613] 7ea0: 00000000 d2027f68 d2027ecc d2027eb8 c092f108 c081be0c c092f0e0 dc6ce900

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.293830] 7ec0: d2027ee4 d2027ed0 c03855c4 c092f0ec 00000004 dc6ce900 d2027f1c d2027ee8

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.302048] 7ee0: c0384c10 c0385580 00000003 00000000 db5fa988 c0384b18 dc788480 bed0d558

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.310265] 7f00: d2027f68 00000000 bed0d558 00000004 d2027f34 d2027f20 c02f9258 c0384b24

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.318482] 7f20: 00000004 dc788480 d2027f64 d2027f38 c02f943c c02f923c 00000000 c031bba0

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.326699] 7f40: c1504dc8 dc788480 00000003 00000000 dc788480 bed0d558 d2027fa4 d2027f68

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.334916] 7f60: c02f96a4 c02f9394 00000003 00000000 d2027fa4 77533b88 c030e758 b6f30620

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.343133] 7f80: b6f30000 00000041 00000004 c01090e4 d2026000 00000000 00000000 d2027fa8

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.351351] 7fa0: c0108f00 c02f9654 b6f30620 b6f30000 00000003 bed0d558 00000004 b6f3061c

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.359568] 7fc0: b6f30620 b6f30000 00000041 00000004 00000000 00000000 004fd000 00000000

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.367785] 7fe0: 00000000 bed0d554 b6ef2363 b6ed13b6 200d0030 00000003 00000000 00000000

Message from syslogd@beaglebone at Oct 18 19:38:42 ...
 kernel:[  222.446978] Code: e5980000 e6ffa07e eb049325 e5985028 (e1d570b0) 
Segmentation fault

Thoughts or additional troubleshooting suggestions appreciated!
Thanks,

mcdeoliveira, 2018-12-22:
I am not sure what the issue is here. I do not own a blue so cannot reproduce your problem. Just a note:

    rcpy_test_motors -s .2

does nothing because the syntax is different. You need

    rcpy_test_motors -s -d .2

for it to set the duty cycle away for zero.

BDD16, 2019-01-25:
I too have the blue....so not sure if it will be of help but thought I'd try.  I'm getting a segmentation fault as well really a bus error warnings.warn('> Robotics cape initialized')
/usr/local/lib/python3.5/dist-packages/rcpy/__init__.py:127: UserWarning: > Installing signal handlers
  warnings.warn('> Installing signal handlers')
Bus error

I've localized that the Bus error is coming from the servo import.  could it be that the pin mapping is not aligned correctly or something to that extent?  If I'm able to develop a fix I'll let you know but in the meantime any hints as to why the servo import causes a Bus error would be cool too!

dmalawey, 2019-03-08:
> I am not sure what the issue is here. I do not own a blue so cannot reproduce your problem. Just a note:
> 

Hi, my team is having the same error come up recurringly.   What can we do to get further support?  We are willing to send you a beaglebone blue or ask Beaglebone.org to send one ASAP.

mcdeoliveira, 2019-03-11:
@dmalawey let me run this by the blue developers. If you email me I can forward it to them.

dmalawey, 2019-03-12:
Bom Dia Mr. Oliveira,

My team needs help in resolving the same issue
<https://github.com/mcdeoliveira/rcpy/issues/14#issue-371687150> described
by Github user MichaelMegliola under your RCPY library.  It's of high
importance because I have a classroom of college students operating mobile
robots that rely on this library.

I'm willing to support the resolution by sending beaglebone blue hardware
if necessary.   We appreciate your assistance with this.

Cheers,

David Malawey, MS ME
Lab Instructor, Texas A&M University

mcdeoliveira, 2019-03-12:
David, I forwarded your issue. May I also recommend that you post that in the beaglebone blue forum?

https://beagleboard.org/discuss#bone_forum_embed

Someone there might have a quick fix for this issue as it seems to be recurrent.

dmalawey, 2019-03-12:
Posted.

https://beagleboard.org/discuss#bone_forum_embed?place=msg%2Fbeaglebone%2FcdywN2pd0_E%2FVkOEB_4MAgAJ

RobertCNelson, 2019-03-12:
@michaelmegliola, can you add this outpu:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

dmalawey, 2019-03-13:
> @michaelmegliola, can you add this output:
> 
> ```
> sudo /opt/scripts/tools/version.sh
> ```
> Regards,

For my device, it is:
```

➜  ~ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBLA21717EL003745]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51                            eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[l                            ocation: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.16.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                            00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190123.0-0rcnee0~stretch+20190123]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                            ers systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk                             weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/m                            mcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.084898] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```

jadonk, 2019-03-24:
@mcdeoliveira how many Blues do you want?

MrRSquared, 2019-05-05:
Hello,
I was just checking in to see if there is any headway with this.  We are having this problem as well.  Is there anything simple we may be overlooking?

MrRSquared, 2019-07-05:
Just as an update, and if anyone else is struggling with this issue, it was fixed.  It is not an rcpy issue, but a librobotcontrol issue. They fixed it on their end. I updated, and think it solved our problem.  [Here is a link to the thread](https://github.com/MXET/SCUTTLE/issues/12).

~Mr. R^2